chipsalliance/f4pga #5: Remove warnings in logs

mgielda, 2019-04-16:
As in title - we need to get rid of those of course, they were ignored temporarily to get this off the ground but now we need to improve.

mithro, 2019-04-16:
Can you paste the warnings in this github issue?

tmichalak, 2019-09-25:
The latest log with the warnings:
<pre>
unning Sphinx v2.2.0
loading pickled environment... done
building [mo]: targets for 0 po files that are out of date
building [html]: targets for 0 source files that are out of date
updating environment: 0 added, 5 changed, 0 removed
reading sources... [100%] vtr-verilog-to-routing/doc/src/dev/index                                                                                                                                                                                                                                                            
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:20: WARNING: toctree glob pattern 'fuzzers/*clb*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:29: WARNING: toctree glob pattern 'fuzzers/*bram*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:38: WARNING: toctree glob pattern 'fuzzers/*iob*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:47: WARNING: toctree glob pattern 'fuzzers/*clk*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:47: WARNING: toctree glob pattern 'fuzzers/*cmt*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:57: WARNING: toctree glob pattern 'fuzzers/*int*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:57: WARNING: toctree glob pattern 'fuzzers/*pip*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:67: WARNING: toctree glob pattern 'fuzzers/*xadc' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/fuzzers/tilegrid'
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/fuzzers/tileconn'
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/fuzzers/ordered_wires'
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/fuzzers/get_counts'
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/fuzzers/dump_all'
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:89: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/fuzzers/timing'
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:98: WARNING: toctree glob pattern 'fuzzers/*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/db_dev_process/parts.rst:109: WARNING: toctree glob pattern 'minitests/*' didn't match any documents
github-symbiflow-docs/source/prjxray/docs/index.rst:26: WARNING: toctree contains reference to nonexisting document 'prjxray/docs/db_dev_process/readme'
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:20: WARNING: toctree glob pattern 'fuzzers/*clb*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:29: WARNING: toctree glob pattern 'fuzzers/*bram*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:38: WARNING: toctree glob pattern 'fuzzers/*iob*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:47: WARNING: toctree glob pattern 'fuzzers/*clk*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:47: WARNING: toctree glob pattern 'fuzzers/*cmt*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:57: WARNING: toctree glob pattern 'fuzzers/*int*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:57: WARNING: toctree glob pattern 'fuzzers/*pip*' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:67: WARNING: toctree glob pattern 'fuzzers/*xadc' didn't match any documents
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/fuzzers/tilegrid'
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/fuzzers/tileconn'
github-symbiflow-docs/source/symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/parts.rst:76: WARNING: toctree contains reference to nonexisting document 'symbiflow-arch-defs/third_party/prjxray/docs/db_dev_process/fuzzers/ordered_wires'
[... cut ...]
writing additional pages...  search/usr/local/lib/python3.7/site-packages/sphinx_symbiflow_theme/search.html:20: RemovedInSphinx30Warning: To modify script_files in the theme is deprecated. Please insert a <script> tag directly in your theme instead.
  {{ super() }}
done
copying downloadable files... [100%] vtr-verilog-to-routing/libs/EXTERNAL/libblifparse/doc/blif.pdf   
copying static files... ... done
copying extra files... done
dumping search index in English (code: en)... done
dumping object inventory... done
build succeeded, 58 warnings.                                        
</pre>
They are spread across several repositories.

mithro, 2019-09-25:
FYI - @acomodi

umarcor, 2022-03-29:
Since the documentation was updated to use intersphinx, instead of submoduling other repos, I believe this issue can be closed.

closed 2022-04-13 by umarcor