chipsalliance/f4pga #195: Docs build exceeds the RAM limit on ReadtheDocs

rw1nkler, 2020-02-26:
The build on ReadtheDocs server exceeds the RAM limit:
https://readthedocs.org/projects/symbiflow/builds/10499598/ 

This is caused by the conda environment and is a well-known issue:
https://github.com/readthedocs/readthedocs.org/issues/6412#issuecomment-560115166

Nevertheless, If we want to use Doxygen on ReadtheDocs we will need conda. Fortunately, In many cases, people from RTD increased the amount of RAM memory for a specific project. Maybe we should ask them for it? 

https://docs.readthedocs.io/en/stable/guides/build-using-too-many-resources.html#requests-more-resources

rw1nkler, 2020-02-28:
It seems that conda itself uses a lot of RAM memory. Here are results of profiling the conda environment creation process:

1) Conda with all channels:

`environment.yml`:
```
name: SymbiFlow-docs
channels:
  - symbiflow
  - conda-forge
  - defaults
dependencies:
  - doxygen
  - cmake
  - pygobject
  - cairo
  - pango
  - pip
  - pip:
    - -r file:requirements.txt
```

[image]

2) Conda only with `conda-forge` channel:

`environment.yml`:
```
name: SymbiFlow-docs
channels:
  - conda-forge
dependencies:
  - doxygen
  - cmake
  - pygobject
  - cairo
  - pango
  - pip
  - pip:
    - -r file:requirements.txt
```

[image]

3) Conda only with pip:

`environment.yml`:

```
name: SymbiFlow-docs
channels:
  - symbiflow
  - conda-forge
  - defaults
dependencies:
  - pip
  - pip:
    - -r file:requirements.txt
```
`requirements.txt`:
```
Sphinx==2.0.0
```
[image]

Profiled using:
```
mprof run --interval 0.01 --include-children ./build/conda/bin/conda env create -q -f environment.yml`
```

rw1nkler, 2020-03-05:
I've asked the RTD team, and we now have 1.5 Gb of RAM for the building process. 
It seems that this is enough because the build is passing. I will close the issue.

closed 2020-03-05 by rw1nkler

mithro, 2020-03-05:
@rw1nkler Did you get the memory increased for all the symbiflow related repos?

rw1nkler, 2020-03-10:
I've asked them only for this repository.