chipsalliance/f4pga #244: VPR documentation does not show up in material theme

daniellimws, 2020-04-14:
The documentation [here](https://symbiflow.readthedocs.io/projects/vtr/en/latest/index.html) shows up in the readthedocs theme, unlike the other symbiflow docs that use the material theme.

umarcor, 2022-03-30:
Since the documentation was updated to use intersphinx, instead of submoduling other repos, I believe this issue can be closed.

VTR does use the readthedocs theme, indeed. However, that's legit, because they build their own docs (https://docs.verilogtorouting.org/) and they decide which theme/style they want to use. Should they want to change the theme, it can be done, but it should be tracked in https://github.com/verilog-to-routing/vtr-verilog-to-routing.

mithro, 2022-04-01:
I think working with the upstream project to use the nicer theme would be good.

umarcor, 2022-04-13:
Closing due to https://github.com/verilog-to-routing/vtr-verilog-to-routing/pull/2006.

closed 2022-04-13 by umarcor