chipsalliance/f4pga #292: VHDL-synthesis using GHDL as option in addition to verific

tmeissner, 2020-06-18:
I think it would be good to mention the possibility to use GHDL as plugin for Yosys as a second, open-source way to use VHDL as RTL language. Would be some additions in the RST and the toolchain-flow.svg image.

Maybe I could create a PR if there is interest.

mithro, 2020-06-18:
@tmeissner - Most certainly be interested in seeing that, much of this was written before the GHDL plugin was viable...

tmeissner, 2020-07-01:
Okay. I will create a PR when I've made the necessary additions to the docs.

umarcor, 2022-10-03:
Although GHDL is not available as a Conda package or supported in f4pga.flows yet, in https://github.com/chipsalliance/f4pga/pull/643 we added CI examples to show how to use a container along with the F4PGA Action in order to generate bitstreams for Xilinx's 7 Series devices from VHDL sources.