chipsalliance/f4pga #395: ModuleNotFoundError: No module named 'breathe' : blocking submodule bumps

tcal-x, 2020-11-25:
@rw1nkler can you take a look -- it looks like automatic bump builds are seeing this error:

```
Traceback (most recent call last):
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow/conda/391/lib/python3.8/site-packages/sphinx/registry.py", line 409, in load_extension
    mod = import_module(extname)
  File "/home/docs/checkouts/readthedocs.org/user_builds/symbiflow/conda/391/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 973, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'breathe'
```

https://github.com/SymbiFlow/symbiflow-docs/pull/391

rw1nkler, 2020-11-25:
Yes I will check what is happening there

rw1nkler, 2020-12-01:
This error is strange. The RTD shows that the necessary packages are installed:
```
Collecting package metadata: ...working... done
Solving environment: ...working... done
Preparing transaction: ...working... done
Verifying transaction: ...working... done
Executing transaction: ...working... done
Ran pip subprocess with arguments:
[u'/home/docs/checkouts/readthedocs.org/user_builds/rw1nkler-symbiflow-docs/conda/fix_submodules/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/docs/checkouts/readthedocs.org/user_builds/rw1nkler-symbiflow-docs/checkouts/fix_submodules/tmpy8J74D.requirements.txt']
Pip subprocess output:
Collecting sphinx_symbiflow_theme
  Cloning http://github.com/SymbiFlow/sphinx_materialdesign_theme.git (to revision master) to /tmp/pip-install-aaj_sxq1/sphinx-symbiflow-theme_787c60f16d1e4b0ea401150d8cf6e6a7
Collecting git+http://github.com/SymbiFlow/sphinx-verilog-domain (from -r file:requirements.txt (line 3))
  Cloning http://github.com/SymbiFlow/sphinx-verilog-domain to /tmp/pip-req-build-ksw9a904
Collecting breathe==4.18.1
  Downloading breathe-4.18.1-py3-none-any.whl (77 kB)
Collecting recommonmark==0.6.0
  Downloading recommonmark-0.6.0-py2.py3-none-any.whl (10 kB)
Collecting Sphinx==3.0.4
  Downloading Sphinx-3.0.4-py3-none-any.whl (2.8 MB)
Requirement already satisfied: setuptools in /home/docs/checkouts/readthedocs.org/user_builds/rw1nkler-symbiflow-docs/conda/fix_submodules/lib/python3.9/site-packages (from Sphinx==3.0.4->-r file:requirements.txt (line 1)) (49.6.0.post20201009)
Collecting sphinx-markdown-tables==0.0.12
  Downloading sphinx_markdown_tables-0.0.12-py3-none-any.whl (15 kB)
Requirement already satisfied: setuptools in /home/docs/checkouts/readthedocs.org/user_builds/rw1nkler-symbiflow-docs/conda/fix_submodules/lib/python3.9/site-packages (from Sphinx==3.0.4->-r file:requirements.txt (line 1)) (49.6.0.post20201009)
Collecting sphinxcontrib-bibtex==0.4.2
  Downloading sphinxcontrib_bibtex-0.4.2-py2.py3-none-any.whl (15 kB)
Collecting sphinxcontrib-domaintools==0.3
  Downloading sphinxcontrib_domaintools-0.3-py3-none-any.whl (6.2 kB)
Collecting sphinxcontrib-images==0.9.2
  Downloading sphinxcontrib_images-0.9.2-py2.py3-none-any.whl (63 kB)
Collecting symbolator==1.0.2
  Downloading symbolator-1.0.2.tar.gz (30 kB)
Collecting alabaster<0.8,>=0.7
  Downloading alabaster-0.7.12-py2.py3-none-any.whl (14 kB)
Collecting babel>=1.3
  Downloading Babel-2.9.0-py2.py3-none-any.whl (8.8 MB)
Collecting commonmark>=0.8.1
  Downloading commonmark-0.9.1-py2.py3-none-any.whl (51 kB)
Collecting docutils>=0.12
  Downloading docutils-0.16-py2.py3-none-any.whl (548 kB)
Collecting hdlparse>=1.0.4
  Downloading hdlparse-1.0.4.tar.gz (10 kB)
Collecting imagesize
  Downloading imagesize-1.2.0-py2.py3-none-any.whl (4.8 kB)
Collecting Jinja2>=2.3
  Downloading Jinja2-2.11.2-py2.py3-none-any.whl (125 kB)
Collecting lark-parser
  Downloading lark_parser-0.11.1-py2.py3-none-any.whl (93 kB)
Collecting markdown==3.0.1
  Downloading Markdown-3.0.1-py2.py3-none-any.whl (89 kB)
Collecting MarkupSafe>=0.23
  Downloading MarkupSafe-1.1.1.tar.gz (19 kB)
Collecting oset>=0.1.3
  Downloading oset-0.1.3.tar.gz (7.3 kB)
Requirement already satisfied: setuptools in /home/docs/checkouts/readthedocs.org/user_builds/rw1nkler-symbiflow-docs/conda/fix_submodules/lib/python3.9/site-packages (from Sphinx==3.0.4->-r file:requirements.txt (line 1)) (49.6.0.post20201009)
Collecting packaging
  Downloading packaging-20.7-py2.py3-none-any.whl (35 kB)
Collecting pybtex>=0.20
  Downloading pybtex-0.23.0-py2.py3-none-any.whl (279 kB)
Collecting latexcodec>=1.0.4
  Downloading latexcodec-2.0.1-py2.py3-none-any.whl (18 kB)
Collecting pybtex-docutils>=0.2.0
  Downloading pybtex_docutils-0.2.2-py2.py3-none-any.whl (4.6 kB)
Collecting Pygments>=2.0
  Downloading Pygments-2.7.2-py3-none-any.whl (948 kB)
Collecting pyparsing>=2.0.2
  Downloading pyparsing-2.4.7-py2.py3-none-any.whl (67 kB)
Collecting pytz>=2015.7
  Downloading pytz-2020.4-py2.py3-none-any.whl (509 kB)
Collecting PyYAML>=3.01
[... 196 lines cut ...]
pip                           20.3
pycairo                       1.19.1
pycparser                     2.20
Pygments                      2.7.2
pyOpenSSL                     20.0.0
pyparsing                     2.4.7
PySocks                       1.7.1
pytz                          2020.4
readthedocs-sphinx-ext        2.1.1
recommonmark                  0.6.0
requests                      2.25.0
setuptools                    50.3.1.post20201107
six                           1.15.0
snowballstemmer               2.0.0
Sphinx                        3.2.1
sphinx-rtd-theme              0.4.3
sphinxcontrib-applehelp       1.0.2
sphinxcontrib-devhelp         1.0.2
sphinxcontrib-htmlhelp        1.0.3
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          1.0.3
sphinxcontrib-serializinghtml 1.1.4
urllib3                       1.25.11
wheel                         0.35.1
```

rw1nkler, 2020-12-01:
Recently there many similar issues reported on the RTD server:
https://github.com/readthedocs/readthedocs.org/issues/7663
https://github.com/readthedocs/readthedocs.org/issues/7665

tmichalak, 2021-07-30:
This seems to be fixed indirectly, dependabot rebases seems to work lately.

closed 2021-07-30 by tmichalak