chipsalliance/f4pga #476: conda env create not working in "Getting Symbiflow"

hansfbaier, 2021-04-13:
```
$ export INSTALL_DIR=/opt/symbiflow

/tmp ⌚ 3:24:59
$ export FPGA_FAM=xc7

/tmp ⌚ 3:25:10
$ bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda;
PREFIX=/opt/symbiflow/xc7/conda
Unpacking payload ...
Collecting package metadata (current_repodata.json): done                                                                     
Solving environment: done

## Package Plan ##

  environment location: /opt/symbiflow/xc7/conda

  added / updated specs:
    - _libgcc_mutex==0.1=main
    - brotlipy==0.7.0=py38h27cfd23_1003
    - ca-certificates==2020.10.14=0
    - certifi==2020.6.20=pyhd3eb1b0_3
    - cffi==1.14.3=py38h261ae71_2
    - chardet==3.0.4=py38h06a4308_1003
    - conda-package-handling==1.7.2=py38h03888b9_0
    - conda==4.9.2=py38h06a4308_0
    - cryptography==3.2.1=py38h3c74f83_1
    - idna==2.10=py_0
    - ld_impl_linux-64==2.33.1=h53a641e_7
    - libedit==3.1.20191231=h14c3975_1
    - libffi==3.3=he6710b0_2
    - libgcc-ng==9.1.0=hdf63c60_0
    - libstdcxx-ng==9.1.0=hdf63c60_0
    - ncurses==6.2=he6710b0_1
    - openssl==1.1.1h=h7b6447c_0
    - pip==20.2.4=py38h06a4308_0
    - pycosat==0.6.3=py38h7b6447c_1
    - pycparser==2.20=py_2
    - pyopenssl==19.1.0=pyhd3eb1b0_1
    - pysocks==1.7.1=py38h06a4308_0
    - python==3.8.5=h7579374_1
    - readline==8.0=h7b6447c_0
    - requests==2.24.0=py_0
    - ruamel_yaml==0.15.87=py38h7b6447c_1
    - setuptools==50.3.1=py38h06a4308_1
    - six==1.15.0=py38h06a4308_0
    - sqlite==3.33.0=h62c20be_0
    - tk==8.6.10=hbc83047_0
    - tqdm==4.51.0=pyhd3eb1b0_0
    - urllib3==1.25.11=py_0
    - wheel==0.35.1=pyhd3eb1b0_0
    - xz==5.2.5=h7b6447c_0
    - yaml==0.2.5=h7b6447c_0
    - zlib==1.2.11=h7b6447c_3

The following NEW packages will be INSTALLED:

  _libgcc_mutex      pkgs/main/linux-64::_libgcc_mutex-0.1-main
  brotlipy           pkgs/main/linux-64::brotlipy-0.7.0-py38h27cfd23_1003
  ca-certificates    pkgs/main/linux-64::ca-certificates-2020.10.14-0
  certifi            pkgs/main/noarch::certifi-2020.6.20-pyhd3eb1b0_3
  cffi               pkgs/main/linux-64::cffi-1.14.3-py38h261ae71_2
  chardet            pkgs/main/linux-64::chardet-3.0.4-py38h06a4308_1003
  conda              pkgs/main/linux-64::conda-4.9.2-py38h06a4308_0
  conda-package-han~ pkgs/main/linux-64::conda-package-handling-1.7.2-py38h03888b9_0
  cryptography       pkgs/main/linux-64::cryptography-3.2.1-py38h3c74f83_1
  idna               pkgs/main/noarch::idna-2.10-py_0
  ld_impl_linux-64   pkgs/main/linux-64::ld_impl_linux-64-2.33.1-h53a641e_7
  libedit            pkgs/main/linux-64::libedit-3.1.20191231-h14c3975_1
  libffi             pkgs/main/linux-64::libffi-3.3-he6710b0_2
  libgcc-ng          pkgs/main/linux-64::libgcc-ng-9.1.0-hdf63c60_0
  libstdcxx-ng       pkgs/main/linux-64::libstdcxx-ng-9.1.0-hdf63c60_0
  ncurses            pkgs/main/linux-64::ncurses-6.2-he6710b0_1
  openssl            pkgs/main/linux-64::openssl-1.1.1h-h7b6447c_0
  pip                pkgs/main/linux-64::pip-20.2.4-py38h06a4308_0
  pycosat            pkgs/main/linux-64::pycosat-0.6.3-py38h7b6447c_1
  pycparser          pkgs/main/noarch::pycparser-2.20-py_2
  pyopenssl          pkgs/main/noarch::pyopenssl-19.1.0-pyhd3eb1b0_1
  pysocks            pkgs/main/linux-64::pysocks-1.7.1-py38h06a4308_0
  python             pkgs/main/linux-64::python-3.8.5-h7579374_1
  readline           pkgs/main/linux-64::readline-8.0-h7b6447c_0
  requests           pkgs/main/noarch::requests-2.24.0-py_0
  ruamel_yaml        pkgs/main/linux-64::ruamel_yaml-0.15.87-py38h7b6447c_1
  setuptools         pkgs/main/linux-64::setuptools-50.3.1-py38h06a4308_1
  six                pkgs/main/linux-64::six-1.15.0-py38h06a4308_0
  sqlite             pkgs/main/linux-64::sqlite-3.33.0-h62c20be_0
  tk                 pkgs/main/linux-64::tk-8.6.10-hbc83047_0
  tqdm               pkgs/main/noarch::tqdm-4.51.0-pyhd3eb1b0_0
  urllib3            pkgs/main/noarch::urllib3-1.25.11-py_0
  wheel              pkgs/main/noarch::wheel-0.35.1-pyhd3eb1b0_0
  xz                 pkgs/main/linux-64::xz-5.2.5-h7b6447c_0
  yaml               pkgs/main/linux-64::yaml-0.2.5-h7b6447c_0
  zlib               pkgs/main/linux-64::zlib-1.2.11-h7b6447c_3

Preparing transaction: done
Executing transaction: done
installation finished.

/tmp ⌚ 3:25:38
$ source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh";

/tmp ⌚ 3:25:46
$ conda env create -f $FPGA_FAM/environment.yml

EnvironmentFileNotFound: '/tmp/xc7/environment.yml' file not found

```
maybe it meant to say this, but it does not work as well:
```
$ conda env create -f $INSTALL_DIR/$FPGA_FAM/environment.yml   

EnvironmentFileNotFound: '/opt/symbiflow/xc7/environment.yml' file not found
```

hansfbaier, 2021-04-13:
I can find an environment.yml here:
`https://github.com/SymbiFlow/symbiflow-docs/issues/476`
maybe it meant to say:
`$ conda env create -f $INSTALL_DIR/$FPGA_FAM/install/environment.yml   `

mithro, 2021-04-13:
@hansfbaier - It would be good to figure out why this isn't failing on GitHub Actions?

mithro, 2021-04-13:
@hansfbaier Also, is this for symbiflow-examples or?

hansfbaier, 2021-04-13:
> @hansfbaier Also, is this for symbiflow-examples or?

Oh yes, so I filed this issue in the wrong project.

hansfbaier, 2021-04-13:
> @hansfbaier - It would be good to figure out why this isn't failing on GitHub Actions?
Can you give me a link to that?

hansfbaier, 2021-04-13:
> I can find an environment.yml here:
> `https://github.com/SymbiFlow/symbiflow-docs/issues/476`
> maybe it meant to say:
> `$ conda env create -f $INSTALL_DIR/$FPGA_FAM/install/environment.yml `
> because this seems to work.

UPDATE, this does not work either:
```
$ conda env create -f $INSTALL_DIR/$FPGA_FAM/install/environment.yml 
Collecting package metadata (repodata.json): done
Solving environment: done

==> WARNING: A newer version of conda exists. <==
  current version: 4.9.2
  latest version: 4.10.0

Please update conda by running

    $ conda update -n base -c defaults conda

Downloading and Extracting Packages
libssh2-1.9.0        | 269 KB    | ################################################################################### | 100% 
libusb-1.0.20        | 100 KB    | ################################################################################### | 100% 
readline-8.1         | 362 KB    | ################################################################################### | 100% 
importlib-metadata-3 | 33 KB     | ################################################################################### | 100% 
make-4.2.1           | 415 KB    | ################################################################################### | 100% 
pycodestyle-2.6.0    | 38 KB     | ################################################################################### | 100% 
prjxray-tools-0.1_28 | 1.2 MB    |                                                                                     |   0% 
prjxray-tools-0.1_28 | 1.2 MB    | ################################################################################### | 100% 
zachjs-sv2v-0.0.5_00 | 1.1 MB    | ################################################################################### | 100% 
icestorm-0.0_0719_g7 | 13.9 MB   | ################################################################################### | 100% 
pcre-8.44            | 212 KB    | ################################################################################### | 100% 
swig-4.0.2           | 1.1 MB    | ################################################################################### | 100% 
mpfr-4.0.2           | 487 KB    | ################################################################################### | 100% 
nodejs-10.13.0       | 13.0 MB   | ################################################################################### | 100% 
expat-2.3.0          | 158 KB    | ################################################################################### | 100% 
icu-58.2             | 10.5 MB   | ################################################################################### | 100% 
prjxray-db-0.0_248_g | 8.5 MB    | ################################################################################### | 100% 
symbiflow-yosys-0.9_ | 10.7 MB   | ################################################################################### | 100% 
libiconv-1.15        | 721 KB    | ################################################################################### | 100% 
certifi-2020.12.5    | 141 KB    | ################################################################################### | 100% 
gcc-riscv64-elf-newl | 61.4 MB   | ################################################################################### | 100% 
libftdi-1.3          | 169 KB    | ################################################################################### | 100% 
bison-3.7.5          | 708 KB    | ################################################################################### | 100% 
zipp-3.4.1           | 15 KB     | ################################################################################### | 100% 
pkg-config-0.29.2    | 245 KB    | ################################################################################### | 100% 
krb5-1.18.2          | 1.3 MB    | ################################################################################### | 100% 
mccabe-0.6.1         | 14 KB     | ################################################################################### | 100% 
openocd-0.10.0_1514_ | 1.7 MB    | ################################################################################### | 100% 
pyflakes-2.2.0       | 56 KB     | ################################################################################### | 100% 
flake8-3.9.0         | 129 KB    | ################################################################################### | 100% 
openjdk-8.0.152      | 57.4 MB   | ################################################################################### | 100% 
typing_extensions-3. | 25 KB     | ################################################################################### | 100% 
capnproto-0.8.0      | 3.6 MB    | ################################################################################### | 100% 
wheel-0.36.2         | 33 KB     | ################################################################################### | 100% 
libuuid-1.0.3        | 15 KB     | ################################################################################### | 100% 
yosys-0.9_5266_g0fb4 | 10.8 MB   | ################################################################################### | 100% 
m4-1.4.18            | 182 KB    | ################################################################################### | 100% 
cmake-3.19.6         | 6.8 MB    | ################################################################################### | 100% 
[... cut ...]
Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: \ Ran pip subprocess with arguments:
['/opt/symbiflow/xc7/conda/envs/symbiflow_arch_def_base/bin/python', '-m', 'pip', 'install', '-U', '-r', '/opt/symbiflow/xc7/install/condaenv.qfksz6va.requirements.txt']
Pip subprocess output:

Pip subprocess error:
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

failed

CondaEnvException: Pip failed
```