chipsalliance/f4pga #513: Rename "Getting Started" section to "Getting started with Developing the SymbiFlow Toolchain"

mithro, 2021-07-23:
People seem to keep ending up at https://symbiflow.readthedocs.io/en/latest/symbiflow-arch-defs/docs/source/getting-started.html and think it is instructions for how to get started with SymbiFlow as a user.

They also ignore the note at the top which says "If you are looking for user documentation, please look at https://github.com/SymbiFlow/symbiflow-examples instead."

I think we need to clearly rename the section to something like "Getting started with development of the SymbiFlow Toolchain" or something.

We should also expand the sections under "Prepare Environment" and "Build example" to explain that you are building the development architecture files and this is expected to take a long time on a fast machine.

mithro, 2021-07-23:
@mgielda - Can you assign someone to look at fixing this as soon as possible?

tmichalak, 2021-07-30:
Fixed with [PR](https://github.com/SymbiFlow/symbiflow-arch-defs/pull/2229)

closed 2021-07-30 by tmichalak