chipsalliance/f4pga #547: Outdated site

goekce, 2022-04-06:
I arrived at the following site which links to this repo and seems to be outdated:

https://symbiflow.readthedocs.io

compared to

https://f4pga.readthedocs.io

Can someone confirm this?

kgugala, 2022-04-06:
Hi @goekce symbiflow webpage is not maintained anymore - you should use f4pga one. Can you tell us where did you find a link to symbiflow? I'd like to update it to point to f4pga

kgugala, 2022-04-06:
We'll also add deprecation warning in symbiflow docs

goekce, 2022-04-06:
- https://symbiflow.github.io/getting-started.html
- click on Symbiflow => https://symbiflow-examples.readthedocs.io/en/latest/
- click on Symbiflow docs => https://symbiflow.readthedocs.io/en/latest

umarcor, 2022-04-28:
Both symbiflow.readthedocs.io and symbiflow-examples.readthedocs.io do now include an "important" admonition telling users about f4pga.org, f4pga.readthedocs.io and f4pga-examples.readthedocs.io. Furthermore, https://github.com/SymbiFlow/.github/blob/master/profile/README.md was added. Therefore, I'm closing this issue.

closed 2022-04-28 by umarcor