chipsalliance/f4pga #561: conda says that my shell has not been configured even after it has been configured

dsw, 2022-05-24:
conda says that by shell has not been configured:

```
f4pga-install$ xc7/conda/bin/conda activate xc7

CommandNotFoundError: Your shell has not been properly configured to use 'conda activate'.
```

but it has been configured:

```
$ grep 'conda initialize' ${HOME}/.bashrc
# >>> conda initialize >>>
# <<< conda initialize <<<
```

I'm running Ubuntu Linux in the default setup (bash).

umarcor, 2022-05-24:
@dsw can you please provide the steps you followed to create the environment? I've seen that happen in CI, but it's typically a transitory issue.

dsw, 2022-05-24:
I attempted to follow your instructions exactly.  This is from my notes.

First I cloned f4pga-examples.

Within f4pga-examples, I ran this:
____
wget https://repo.continuum.io/miniconda/Miniconda3-latest-Linux-x86_64.sh
-O conda_installer.sh
____
Which exited with success.

Within f4pga-examples I ran this:
____
export INSTALL_DIR=${HOME}/f4pga-install
export FPGA_FAM=xc7
bash conda_installer.sh -u -b -p $INSTALL_DIR/$FPGA_FAM/conda;
source "$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh";
conda env create -f $FPGA_FAM/environment.yml
mkdir -p $INSTALL_DIR/xc7/install
for PKG in install xc7a50t_test xc7a100t_test xc7a200t_test xc7z010_test; do
  wget -qO-
https://storage.googleapis.com/symbiflow-arch-defs/artifacts/prod/foss-fpga-tools/symbiflow-arch-defs/continuous/install/20220406-185509/symbiflow-arch-defs-${PKG}-3ef4188.tar.xz
| tar -xJC $INSTALL_DIR/xc7/install
done
____
Which exited with success.

Note that right now for other reasons I need to run an older version of
Ubuntu, so I am still running Ubuntu 14.04.

dsw, 2022-05-24:
> After installing conda, did you restart your shell?

You mean after it modified my .bashrc ?  Yes, I logged out and logged in
again.

dsw, 2022-05-24:
Here are the changes to my .bashrc (modulo some deletions marked
with ellipses):
____
# this is for f4pga:

F4PGA_INSTALL_DIR=${HOME}/f4pga-install
F4PGA_FPGA_FAM=xc7
export PATH="$F4PGA_INSTALL_DIR/$F4PGA_FPGA_FAM/install/bin:$PATH";

# >>> conda initialize >>>

# !! Contents within this block are managed by 'conda init' !!

__conda_setup="$('[. . . home dir omitted . .
.]/f4pga-install/xc7/conda/bin/conda' 'shell.bash' 'hook' 2> /dev/null)"
if [ $? -eq 0 ]; then
    eval "$__conda_setup"
else
   [. . . omitted . . .]
fi
unset __conda_setup
# <<< conda initialize <<<
____

I also notice that my bash prompt string has changed and is now preceded by
the string "(base) " before the usual ***@***.***:directory" prompt
string.  Given that something changed, I think the .bashrc is running.

dsw, 2022-05-24:
Seeing your changes to the repo, I cloned f4pga-examples from scratch
and followed the instructions again.  Same problem.

umarcor, 2022-05-24:
@dsw, I could reproduce the issue in CI (https://github.com/antmicro/f4pga/actions/runs/2380264724), using ubuntu 18.04, 20.04 or 22.04. However, I did find the solution yet.

umarcor, 2022-05-24:
@dsw, see https://github.com/antmicro/f4pga/commit/0696ecf8744de61ee582353967a0524a709c31b0. 

According to https://f4pga-examples.readthedocs.io/en/latest/building-examples.html, you need to both export the path and source `...profile.d/conda.sh`.

dsw, 2022-05-25:
Ok, sorry about that.  When I read your install instructions say to source
"$INSTALL_DIR/$FPGA_FAM/conda/etc/profile.d/conda.sh" I assumed that this
was the documentation being unclear and that should be done at install, not
every time I log in, as the most configuration I have ever seen in a
.bashrc is to set environment variables.  I must say, this whole setup is
rather non-standard, but I suppose you find it natural.  Thanks for
explaining.

umarcor, 2022-05-25:
@dsw, I acknowledge that the setup is non-standard. Without going into much detail, It's a mix of using Conda, having the FPGA_FAM envvar (because of differences between the xc7 and quicklogic families), and "manually" installing the arch-defs packages. All of that is being worked on:

- There differences in the layout of arch-defs packages for xc7 and quicklogic is being homogeneized.
- Last week, Python package `f4pga` was merged in this repo (https://github.com/chipsalliance/f4pga/tree/main/f4pga). That will eventually allow to get rid of the F4PGA_FAM and the INSTALL_DIR envvars, replacing them with CLI options (or a JSON configuration file).
- Eventually, enough tools and patches will be upstreamed to the repos of most known distributions, and Conda won't be a requirement. Meanwhile, having the F4PGA Conda environment preinstalled in a container is a kind of cheating, but does works:
  - https://hdl.github.io/containers/ToolsAndImages.html#f4pga-conda
  - https://hdl.github.io/containers/ug/AllInOne.html#f4pga

closed 2022-05-25 by umarcor

dsw, 2022-05-25:
It might help to say something like "yes, we know it is very weird to
source this script from your .bashrc, but that is in fact what we want you
to do".  I mean, you did say it and I did not do what it said, but it might
help to point it out.

Also, as far as I can tell you do not need to export INSTALL_DIR and it is
namespace pollution to do so: recall that exported env vars are
system *global* and therefore in any project I maintain I always pick some
three or four letter prefix for the whole project and prefix *all* related
env vars with it in order to avoid namespace collisions.

umarcor, 2022-05-25:
@dsw, it's work in progress. See https://github.com/chipsalliance/f4pga-examples/pull/306/files and https://github.com/chipsalliance/f4pga/issues/562.

dsw, 2022-05-25:
Would you like me to upload my collection of all of your instructions
expressed as two files: an install script and a .bashrc insert?  It seems
like a rather minor contribution, as I just copied it from your web page,
but I think it would have simplified my thinking quite a bit and saved me
some time and errors, even though the errors were my fault for not reading
the docs carefully.  If so, just tell me where you want it and I'll add it
and send a pull request.  They do actually work :-).

umarcor, 2022-05-25:
@dsw, actuallly, those code-blocks are treated as "executable snippets" by the CI. "tuttest" parses the sources of the docs, extracts code-blocks and executes them. See https://github.com/chipsalliance/f4pga-examples/blob/main/.github/scripts/install-toolchain.sh#L48 and https://raw.githubusercontent.com/chipsalliance/f4pga-examples/main/docs/getting.rst. Therefore, I don't know how would contributing the "standalone" scripts play with having them tested in CI. Maybe "including" them in rst does work, but I'm unsure.

Nevertheless, you might want to add an "admonition" to the documentation, to provide a "hint" to future readers. There, you might suggest them to save those lines in files and add them to `.bashrc`. The sources are:

- Getting: https://github.com/chipsalliance/f4pga-examples/blob/main/docs/getting.rst
- Building examples: https://github.com/chipsalliance/f4pga-examples/blob/main/docs/building-examples.rst

umarcor, 2022-05-25:
The syntax to add admonitions is the following:

```rst
.. HINT::
  Your text goes here.
  
  You can include code-blocks if you want:
  
  .. code-block::
  
    source myscript.sh
```