chipsalliance/f4pga #589: https://symbiflow-examples.readthedocs.io/ is still online

tcal-x, 2022-07-29:
https://symbiflow-examples.readthedocs.io/en/latest/getting.html is still online, and is not up to date with the [f4pga version](https://f4pga-examples.readthedocs.io/en/latest/getting.html).

mithro, 2022-07-29:
We should make sure that https://symbiflow-examples.readthedocs.io redirects to the f4pga version using ReadTheDocs redirect support https://docs.readthedocs.io/en/stable/user-defined-redirects.html

mithro, 2022-07-29:
@kgugala / @mgielda / @umarcor - Any chance one of you can look at making this happen?

mithro, 2022-09-24:
Poke?

kgugala, 2022-09-26:
hmm, the redirect is set (exactly as the docs say), but it doesn't seem to be working. One option here is to remove the contents of the docs hosted on symbiflow-examples webpage and have only one page with information that the project has been moved.

Other one is to inject a script to the theme that will do the redirect. What do you think?

umarcor, 2022-09-26:
We looked into this some months ago. The redirect feature of RTD does not work for us, because of various reasons. The main one is that it's mostly meant for statically redirecting the pages of a site that was moved to a different host/URL or to some subdir. However, in our case the content was reorganised, split and moved around. Therefore, there is no 1-to-1 match that we can use to redirect old URLs (pages). We could break (remove) all the pages and have all of them redirected to the index of the new site, but that would probably require manual action by RTD admins (*"This option is only available on some plan levels"*), or doing the redirection in JS/HTML as suggested by @kgugala. We also tried changing the URL of the project, rather than creating a new project/site; that's unfortunately not possible.

Hence, we decided to add an "Important" admonition at the top of each page in symbiflow.rtfd.io and symbiflow-examples.rtfd.io (through a prolog; see https://github.com/f4pga/f4pga/commits/symbiflow and https://github.com/chipsalliance/f4pga-examples/commits/symbiflow), to tell users that "SymbiFlow is now F4PGA" and to point them to the new locations: https://f4pga.org/, https://f4pga.readthedocs.io/ and https://f4pga-examples.readthedocs.io/.