chipsalliance/f4pga #648: Build failure on litex SoC for Arty with Symbiflow

alanvgreen, 2022-10-29:
I have installed Symbiflow following the instructions at https://f4pga-examples.readthedocs.io/en/latest/getting.html

With the conda environment active, I installed LiteX using the litex_setup.py script as perhttps://github.com/enjoy-digital/litex#quick-start-guide

I then ran `python -m litex_boards.targets.digilent_arty --toolchain symbiflow --build `. I expected to get a working bitstream but instead I got a failure from VPR.

The final lines of output were. 

```
Found constant-zero generator '$abc$82080$auto$blifparse.cc:518:parse_blif$89937.T0'
# Load circuit took 0.35 seconds (max_rss 90.7 MiB, delta_rss +32.2 MiB)
The entire flow of VPR took 1.21 seconds (max_rss 90.7 MiB)

stderr:
Error 1: 
Type: Blif file
File: /home/osboxes/src/synth/build/digilent_arty/gateware/digilent_arty.eblif
Line: 184330
Message: Failed to find matching architecture model for 'DNA_PORT'
```

There is a `DNA_PORT` cell in the generated Verilog

```
DNA_PORT DNA_PORT(
	.CLK(dna_clk),
	.DIN(1'd0),
	.READ((main_dna_count == 1'd0)),
	.SHIFT(1'd1),
	.DOUT(main_dna_dout)
);
```

This issue seems similar to https://github.com/f4pga/f4pga-arch-defs/issues/2123, but looking at the entry for DNA_PORT in cells_xtra.v, all seems well. There was a recent-ish change to how LiteX uses DNA_PORT, (https://github.com/enjoy-digital/litex/issues/1411,  [b24d744f8ed](https://github.com/enjoy-digital/litex/commit/b24d744f8ed553c023e6ab54e05d237de3dadbb7)) though, to my untrained eye, there is nothing suspicious about that change.

alanvgreen, 2022-10-29:
This design synthesizes without errors using the Vivado toolchain.

mithro, 2022-10-29:
@kgugala - Can you get someone to look at this?

alanvgreen, 2022-11-11:
Hi @kgugala! Unfortunately, the litex-boards definition for the vanilla Arty board includes DNA by default, with no way to turn it off: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/digilent_arty.py#L97

I'm thinking that either F4PGA should support DNA, or the Arty target in litex-boards shouldn't enable it by default. What's the right way to go here?

pgielda, 2022-11-11:
Hi Alan,

We will take a look at how hard it is to add DNA to the toolchain.

mlaga97, 2023-01-09:
I believe that this is now resolved in litex-boards.

closed 2023-03-03 by umarcor