chipsalliance/f4pga #655: Unable to install environment in WSL1, Ubuntu 20.04

JsGraaf, 2023-02-10:
When running `conda env create -f $FPGA_FAM/environment.yml` I get the following error:
`Collecting package metadata (repodata.json): done
Solving environment: done

Downloading and Extracting Packages

Preparing transaction: done
Verifying transaction: done
Executing transaction: done
Installing pip dependencies: / Ran pip subprocess with arguments:
['/home/joris/.conda/envs/eos-s3/bin/python', '-m', 'pip', 'install', '-U', '-r', '/home/joris/Scriptie/Toolchain/F4PGA/f4pga-examples/eos-s3/condaenv.ciz9ugbc.requirements.txt', '--exists-action=b']
Pip subprocess output:
Collecting https://github.com/chipsalliance/f4pga/archive/e1cd038f06c7161b27afd0073fb507da2b8e5a9e.zip#subdirectory=f4pga (from -r /home/joris/Scriptie/Toolchain/F4PGA/f4pga-examples/eos-s3/../common/requirements.txt (line 18))
  Using cached https://github.com/chipsalliance/f4pga/archive/e1cd038f06c7161b27afd0073fb507da2b8e5a9e.zip
  Installing build dependencies: started
  Installing build dependencies: finished with status 'done'
  Getting requirements to build wheel: started
  Getting requirements to build wheel: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting git+https://github.com/antmicro/quicklogic-fasm.git (from -r /home/joris/Scriptie/Toolchain/F4PGA/f4pga-examples/eos-s3/requirements.txt (line 20))
  Cloning https://github.com/antmicro/quicklogic-fasm.git to /tmp/pip-req-build-yrrq_zg4

Pip subprocess error:
  Running command git clone --filter=blob:none --quiet https://github.com/antmicro/quicklogic-fasm.git /tmp/pip-req-build-yrrq_zg4
  fatal: unable to access 'https://github.com/antmicro/quicklogic-fasm.git/': Could not resolve host: github.com
  error: subprocess-exited-with-error

  × git clone --filter=blob:none --quiet https://github.com/antmicro/quicklogic-fasm.git /tmp/pip-req-build-yrrq_zg4 did not run successfully.
  │ exit code: 128
  ╰─> See above for output.

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× git clone --filter=blob:none --quiet https://github.com/antmicro/quicklogic-fasm.git /tmp/pip-req-build-yrrq_zg4 did not run successfully.
│ exit code: 128
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip.

failed

CondaEnvException: Pip failed`

I have already tried changing my git global config to not use a proxy. 

When ignoring the error and starting the environment, when compiling the btn_counter example, I get the error f4pga: command not found.

Thanks in advance for any help!

pgielda, 2023-02-10:
Maybe I am missing something in this log but "Could not resolve host: github.com" might suggest some DNS problems on your side?

JsGraaf, 2023-02-11:
However, I can ping Github just fine and download different packages. I only receive the error from the conda command.

JsGraaf, 2023-02-13:
This turned out to be a WSL1 issue, compilation works fine in WSL2

closed 2023-02-13 by JsGraaf